TonyTromp/tidal-connect-docker #6: Volume controlls

lomdar67, 2022-02-02:
Probably not a issue just me being stupid:

NO VOLUME * Please check your volume setting on your device and use your device to increase the volume. If setup/dac is recognized it you will see volume changes also updating in the HifiBerry Audio Controls.

If I understand it correctly volume control with e.g. the iOS or Android Tidal app should be working, right?

I have a HIFIBERRY DAC2 PRO with a Raspberry 3B+ running HifiBerryOS (latest version). I've installed the latest version of the docker image (yesterday). So far Tidal Connect is working very well, but I can't control  volume. Is there anything I can try or do?

Best regards,
Michael

TonyTromp, 2022-02-02:
> Probably not a issue just me being stupid:
>
> NO VOLUME * Please check your volume setting on your device and use your device to increase the volume. If setup/dac is recognized it you will see volume [...]

Sad to see the volume control is still not working on the HIFIBERRY DAC2 PRO. yes, volume control should be working via the mobile IOS/Android device.

Can you please help me debug this issue as i dont own a HIFIBERRY DAC2 PRO 
1. Please pull/download the latest version
2. Install using the installation script. During installation it will ask for a playback device (typically your DAC is the first choice '0')
3. Open the speaker-controller application using the script: ./show-speaker-controller
4. Try and play a song from Tidal using your mobile device
5. In the UI terminal check if the speaker_controller app does anything for you e.g. 'P' for play/pause -/+ for Volume controls
6. If the above works: try with your mobile IOS/Android device to change the volume
7. IMPORTANT: Exit the Application using tmux key-combination: ctrl+b, d (detach)

Repeat steps 2-7 with a different audio device (maybe audio mixer or sysdefault works, i am curious to know).

Hope this helps you both.

Thanks,
Tony

jrotaetxe, 2022-02-02:
> Sad to see the volume control is still not working on the HIFIBERRY DAC2 PRO. yes, volume control should be working via the mobile IOS/Android device.
>
> Can you please help me debug this issue as i [...]

Sure, Tony. And let me say I am a pig in the mud with the work you have done as is it now.... Thank you very much.
The UI shows complete control. Says the artist, the state... everything. If I touch the volume control, the bars with ### become narrower or wider, so the commands ate catched by the speaker controller app... The only issue is that those controls have no effect
...
[image]
[image]

lomdar67, 2022-02-02:
Hello Tony,

These are the devices that show up:

0=snd_rpi_hifiberry_dacplus: HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0 (hw:0,0)
1=sysdefault
2=default
3=dmix

For device 0:

P,L,K do work
When I press + or - I see, that the volume in the Tidal iPhone app and speaker_controller changes, but the volume I hear doesn't change at all. 

For device 1:
same as device 0

For device 2:
same as device 0

When I select device 3 I see Tidal Connect in the app and I can connect, but I can't play anything.

Hope that helps.

Best reagards,
Michael

TonyTromp, 2022-02-02:
It seems that speaker_application works as expected (and controls are captured correctly), but the commands are now trickled down to the DAC volume / DAC device for some odd reason. I will need to research the drivers on this device, but will be tricky as i have no way of testing the HIFIBERRY DAC2 PRO. 

Question.
What is your dtoverlay for your DAC in /boot/config.txt ?

this is mine (and hifiberry-dac is the one to use even that i have a DAC+DSP)
```
dtoverlay=vc4-fkms-v3d,audio=off
dtoverlay=hifiberry-dac
```

Would you be able to change the dtoverlay and try if those settings work?

jrotaetxe, 2022-02-02:
[image]
Will check with yours...

lomdar67, 2022-02-02:
> It seems that speaker_application works as expected (and controls are captured correctly), but the commands are now trickled down to the DAC volume / DAC device for some odd reason. I will need to [...]

This is strange. I didn't have a overlay for the DAC:

`# Please note that this is only a sample, we recommend you to change it
to fit # your needs.
# You should override this file using a post-build script.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

# We always use the same names, the real used variant is selected by
# BR2_PACKAGE_RPI_FIRMWARE_{DEFAULT,X,CD} choice
start_file=start.elf
fixup_file=fixup.dat

kernel=zImage

# To use an external initramfs file
#initramfs rootfs.cpio.gz

# Disable overscan assuming the display supports displaying the full
resolution # If the text shown on the screen disappears off the edge,
comment this out disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=100
gpu_mem_512=100
gpu_mem_1024=100
# Enable I2C and SPI
dtparam=i2c=on
dtparam=spi=on
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1
dtoverlay=vc4-fkms-v3d,audio=off`

I added you overlay and did a reboot, but the behavior is exactly the same as bevore.

`dtoverlay=hifiberry-dac`

lomdar67, 2022-02-02:
> Will check with yours... mmmm. Config txt is R/O. Have to extract the memory card to edit from outside. Will try tomorrow

Just remount the partition as read/write:

`mount -o remount,rw /dev/mmcblk0p1 /boot`

TonyTromp, 2022-02-03:
I use ``` mount -o remount,rw /boot ``` which works similarly as described above.

Also i find a setting that could be related to our issue, as i read about HIFIBERRY DAC2 having a Hardware Volume option.

Can you execute the following command:
```
/opt/hifiberry/bin/alsa-mode
```
This returns SOFTVOL for me and i am curious if its different on your device.

TonyTromp, 2022-02-03:
btw very strange that you dont have a overlay for your DAC?
For me the settings were added automatically.

Did you have a look at: https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/ ?

lomdar67, 2022-02-03:
> I use `mount -o remount,rw /boot` which works similarly as described above.
>
> Also i find a setting that could be related to our issue, as i read about HIFIBERRY DAC2 having a Hardware Volume [...]

It also return  SOFTVOL:

```
# /opt/hifiberry/bin/alsa-mode
SOFTVOL
#
```

lomdar67, 2022-02-03:
> /opt/hifiberry/bin/fix-dacadcpro-mixer

Yes, the DAC was working without the overlay.

I added dtoverlay=hifiberry-dacplus and this gets not changed after reboot.

alsa-mode is still SOFTVOL after reboot, no output from fix-dacadcpro-mixer and here is amixer:

```
# /opt/hifiberry/bin/alsa-mode
SOFTVOL
# /opt/hifiberry/bin/fix-dacadcpro-mixer
# amixer
Simple mixer control 'Headphone',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 63
  Mono: Playback 54 [86%] [0.00dB]
Simple mixer control 'DSP Program',0
  Capabilities: enum
  Items: 'FIR interpolation with de-emphasis' 'Low latency IIR with de-emphasis'                                                                  'High attenuation with de-emphasis' 'Fixed process flow' 'Ringing-less low late                                                                 ncy FIR'
  Item0: 'FIR interpolation with de-emphasis'
Simple mixer control 'DSPVolume',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]
Simple mixer control 'Analogue',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 1
  Mono:
  Front Left: Playback 1 [100%] [0.00dB]
  Front Right: Playback 1 [100%] [0.00dB]
Simple mixer control 'Analogue Playback Boost',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 1
  Front Left: 0 [0%] [0.00dB]
  Front Right: 0 [0%] [0.00dB]
Simple mixer control 'Auto Mute',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Auto Mute Mono',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Auto Mute Time Left',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Auto Mute Time Right',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Clock Missing Period',0
  Capabilities: enum
  Items: '1s' '2s' '3s' '4s' '5s' '6s' '7s' '8s'
  Item0: '1s'
Simple mixer control 'Deemphasis',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Digital',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 207
  Mono:
  Front Left: Playback 207 [100%] [0.00dB] [on]
  Front Right: Playback 207 [100%] [0.00dB] [on]
Simple mixer control 'Max Overclock DAC',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock DSP',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock PLL',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 20
  Mono: 0 [0%]
Simple mixer control 'Softvol',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]
Simple mixer control 'Volume Ramp Down Emergency Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Emergency Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '4dB/step'
Simple mixer control 'Volume Ramp Down Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'
Simple mixer control 'Volume Ramp Up Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Up Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'
```

```
# /opt/hifiberry/bin/alsa-mode
SOFTVOL
#
```

jrotaetxe, 2022-02-03:
Hi. My results are consistent with those presented by lomdar67
Regards

TonyTromp, 2022-02-03:
ok that is good to know. For now i am a bit in the dark (i first thought maybe the SOFTVOL was different but seems the same). Since i didnt create the speaker_application it will be hard to modify, so hope i can create a workaround.

I have added in the README that HIFIBERRY DAC PLUS volume control doesnt work as of now. Hope we can find a fix soon.

Python21212, 2022-02-08:
Hi there, has there been any update to this problem? I'm considering getting a DAC2 Pro but it might not be a good idea if it's very inconvenient to change the volume.

TonyTromp, 2022-02-09:
If you want Volume control via iOS i can not recommend this. You might have more luck with another OS/Tidal but almost all tidal connect raspberry implementations are based of the same binaries found on github. Since we never get a opensource tidal binary, it will be hard to implement. maybe one day there will be a official implementation of HifiberryOS, but right now this really needs help of the community as it goes beyond my reach.

kls0e, 2022-03-19:
i use the great tidal-connect-docker in combo with an RPI3B+ and a `Hifiberry AMP+`. It works fine! 

My only request is that the volume control within the tidal-connect app should control `softvol` rather than `master` volume.  Is the essence of this thread that this is not possible with the Tidal binaries? Or is there a way?
[image]

Akfree1st, 2022-04-23:
Hi.
I use the normal Hifiberry-dac and also have no control over the volume.
I have to turn my amp to nearly zero befor starting Tidal playback because ist default at 100% or more no matter witch volume was played with other integrations bevor.
Just for your information.

I use the latest version of HifiBerryOS, a RPi4 and a hifiberry-dac

pfbreton-perso, 2022-06-17:
_I have to turn my amp to nearly zero befor starting Tidal playback because ist default at 100% or more no matter witch volume was played with other integrations bevor_

Same here.  If I could reduce the default volume by lets say 50% even if its a set volume, that would be better.  For some reason, the output voltage of the DAC is really high, forcing the preamp volume control to a very very low setting. So we lose granual volume control.

jerver, 2022-08-11:
Hi, Same with Hifiberry DIGI+. No working volume control on phone and show-speaker-controller. Is this the same issue or am I doing something else wrong.

jfk22, 2023-07-31:
I can confirm the same problem with the Hifiberry MiniAmp. Playing is working fine, but volume control is working neither via app nor via HifiberryOS web interface.

MDrueke, 2023-10-03:
First of all thank you so much for this great docker tidal connect implementation!
Just wanted to mention that on volumio the iOS volume control works  (Tidal connect with DAC+). Maybe it's encouraging to know that it is possible!