TonyTromp/tidal-connect-docker #7: wip - UI metadata doesnt update when running audiocontrol2 as service.

TonyTromp, 2022-02-02:
For some odd reason running audiocontrol2 in foreground works fine, but when running as daemon/service (systemctl start audiocontrol2) it doesnt pick up any changes.

charliesjc, 2022-07-03:
I have made some headway with this and I think it's exactly how the communication is supposed to happen. Combing the log files in level 6 (TRACE) there is a process called "tisoc" controlling everything. I found a unix socket at /tmp/tisoc-controller and connected to it with socat. It shows all the inbound/outbound JSON comms that I assume you're supposed to read for metadata info and send commands to for control.

I haven't been able to get much further than see what's happening because once you connect via socat, the file basically gets locked and my phone can't connect to Tidal-Connect and/or the speaker controller can't connect either. So I have to restart the Tidal-connect service. BUT, if you connect with your phone first, then use socat, you can see all the progress, pause/play/next/previous from the phone and the notifications for metadata changes with all the track/playlist details. This is obviously the same info as in the detailed logs but it means you can interface with it this way.

There is also a debugging port that you can turn on and it creates a FIFO socket at /tmp/scdport but after opening it with socat I haven't got a clue what to do with it.

I've been trying to work out how to "view" the tisoc-controller without blocking it so you can see which commands are sent from the speaker controller/other apps. I have confirmed that this seems to be the way to interface with it because if you do the same socat on the tisoc-controller on the other raspberry music integration that you have to pay for, for Tidal-connect, it breaks in exactly the same way and the interface becomes useless without a restart of TC.

charliesjc, 2022-08-16:
I think this may be dead in the water. I've noticed that with the latest update to the Android Tidal app, v 2.65.0, connecting to the tidal-connect service fails with a TLS handshake problem. It works fine on Volumio, presumably because they've received updated certificates.

MoM-Raider, 2022-08-17:
Jupp, after the update to Android Tidal 2.65.1 the App does not connect any more...

charliesjc, 2022-08-17:
> Jupp, after the update to Android Tidal 2.65.1 the App does not connect any more...
> 

Sigh 😔

TonyTromp, 2023-02-11:
> > Jupp, after the update to Android Tidal 2.65.1 the App does not connect any more...
> 
> Sigh 😔

https://github.com/TonyTromp/tidal-connect-docker/issues/28 please follow this issue on this thread. This issue is meant for UI development.

nasawyer7, 2024-07-02:
Is there any way to get now playing to display any song information now when using tidal connect?