TonyTromp/tidal-connect-docker #8: Tidal Connect invisible in any devices - no Hifiberry on list available devices to connect

Zycho-pl, 2022-02-03:
1. Did you run the ./install_hifiberry script? - yes
2. Which playback device did you select? - 0 (chosen in installation process)
devices list:
device#0=snd_rpi_hifiberry_dacplushd: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 (hw:0,0)
device#1=sysdefault
device#2=default
device#3=dmix

3. if you type 'docker ps' do you see tidalconnect container running? - yes
[image]

4. what Hifiberry DAC do you use? - DAC 2HD

TonyTromp, 2022-02-03:
Hi Zycho,

Seems that there is something wrong with the setup (restarting only happens when something is wrong).
Could be a type in your entrypoint.sh file.

Please try the following.
1.  cd Docker
2. docker-compose down
3. then delete the while tidal-connect folder and re-install (e.g. cd .. && rm -Rf tidal-connect-docker

After you re-installed: try the following
1. cd Docker
2. docker-compose down
3. docker-compose up
This will run the command in the Foreground, so you can see why its 'Exiting'. If its not exiting, its working and you can restart (ctrl + c and then run: docker-compose down && docker-compose up -d).

Hope this helps.

Kind regards,
Tony

Zycho-pl, 2022-02-04:
Hi Tony,

Thank you for your support. I reinstalled Tidal Connect docker by your advice, but after that i couldn't start the docker service. So then i decided to reinstall Hifiberry OS and install Tidal Connect docker again. So now everything is working fine. But i noticed one strange issue - after formatting my sd card twice and strating yor script i had en error - that tidal-connect directory already exist (even if i formatterd  my sd card twice). But after removing existing directory and running script again, everything installed and started propperly. So I suppose it could be problem with my sd card.
Anayway, many thanks for your work and support, especially because Tidal Connect is one of the most important functionality for me.

Best regards,
Piotr

TonyTromp, 2022-02-05:
Ok. that is certainly wierd and i will continue to investigate on this. Glad it now works.
I think the installation script does not 'update' the symbolic link to another tidal-docker-installation.
Will update the script ith this input later this weekend.

Thanks for your support!

Kind regards,
Tony

SamPT81, 2022-02-06:
Hi Toni. First of all thank you for making this "plugin" available!

I am also having the same issues that Zycho has. Even though I have it installed and runing (can confirm it both via command line and via the UI), hifiberry device is not visible to be "connected to".

Set-up:
RP 3+ && DAC 2 HD (same as Zycho)

TonyTromp, 2022-02-07:
Hi Sam,

Thanks for your feedback.
Can you and run the script, from the tidal-docker-connect folder

./select-playback-device 

Then select your audio device and try again running (from the Docker subfolder)

docker-compose down &&  docker-compose up -d

Kind regards,
Tony

Seetroll, 2022-02-11:
Hi Tony & all
thanks so much for providing this installation script.
I had the same issue as Zycho and Sam. I found a workarround with manualy exporting the HiFi Berry device No.
     export PLAYBACK_DEVIDE=0
     docker-compose down && docker-compose up -d
It seems to be a problem with the script 'select_playback_device'. After that all works fine. 
It's only a workarround which solves the problem only temporarely.
Anyway thank you for your work.
Seetroll

Seetroll, 2022-02-12:
The script 'select_playback_device' can't find any implemented devices and leaves PLAYBACK_DEVICE undefined.

TonyTromp, 2022-02-14:
Hi @Seetroll , Can you double check if your docker container is running?

``` docker ps ```