TonyTromp/tidal-connect-docker #12: Audio output setting lost after Raspberry restart

jmayday, 2022-02-23:
After I restarted my Raspberry, HiFiBerry stopped working. Well, it was working but not pushing any output to my target device. I have impression that "audio output" setting has been lost. 

I tried that script but it didn't even list any audio outputs:
```
# ./select-playback-device

Found output devices...

Choose your output Device (0-9): 0

Playback device is set to:
# ./install_hifiberry.sh
```
I had to run install_hifiberry.sh script in order to fix the problem:
```
# ./install_hifiberry.sh
Running environment:
  FRIENDLY_NAME:            hifiberry
  MODEL_NAME:               hifiberry
  BEOCREATE_SYMLINK_FOLDER: /opt/beocreate/beo-extensions/tidal
  DOCKER_DNS:               8.8.8.8
  DOCKER_IMAGE:             edgecrush3r/tidal-connect:latest
  BUILD_OR_PULL:            pull
  MQA_PASSTHROUGH:          false
  MQA_CODEC:                false
  PWD:                      /data/tidal-connect-docker

Wed Feb 23 16:59:51 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Pre-flight checks.
Wed Feb 23 16:59:51 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Checking to see if Docker is running.
Wed Feb 23 16:59:52 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Confirmed that Docker daemon is running.
Wed Feb 23 16:59:52 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Checking to see if Docker image edgecrush3r/tidal-connect:latest exists.
Wed Feb 23 16:59:52 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Docker image edgecrush3r/tidal-connect:latest exist on the local machine.
Wed Feb 23 16:59:52 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Stopping Tidal Container..

Stopping TIDAL Connect Service.
Finished stopping TIDAL Connect Service.
Wed Feb 23 17:00:08 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Select audio output device

Found output devices...

0=snd_rpi_hifiberry_dac: HifiBerry DAC HiFi pcm5102a-hifi-0 (hw:0,0)
1=ADI-2 DAC (51090305): USB Audio (hw:1,0)
2=sysdefault
3=default
4=dmix
Choose your output Device (0-9): 1
ADI-2 DAC (51090305): USB Audio (hw:1,0)
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Creating .env file.
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Finished creating .env file.
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: /data/tidal-connect-docker/Docker/CONFIG already exists. this file will be replaced with new configuration.
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Create config symlink -> /data/tidal-connect-docker/Docker/.env
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Generating docker-compose.yml.
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Finished generating docker-compose.yml.
Wed Feb 23 17:00:16 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Enabling TIDAL Connect Service.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Finished enabling TIDAL Connect Service.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Adding TIDAL Connect Source to Beocreate.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: TIDAL Connect extension found, removing previous install.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Adding TIDAL Connect Source to Beocreate UI.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Finished adding TIDAL Connect Source to Beocreate.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Installation Completed.
Wed Feb 23 17:00:17 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Starting TIDAL Connect Service.

Starting TIDAL Connect Service.
Finished starting TIDAL Connect Service.
Wed Feb 23 17:00:24 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Restarting Beocreate 2 Service.
Stopping Beocreate 2 Server
Starting Beocreate 2 Server
Finished restarting Beocreate 2 Server.
Wed Feb 23 17:00:28 CET 2022 hifiberry install_hifiberry.sh[2224]: [INFO]: Finished, exiting.
#
```

Could you please advise how to make whole configuration durable?

TonyTromp, 2022-03-02:
Not sure what your goal is but the ./select-playback-device can only be run if the docker image is started. 
Run the ./install_hifiberry.sh script should always be the first step.

jmayday, 2022-03-07:
Well, yes this is what I would expect that it survives reboots. Running ./install_hifiberry.sh did solve the problem (as visible on console output in my previous post) but running it every time config is lost would be somehow problematic. Is there a way to troubleshoot it when issue occurs again?