TonyTromp/tidal-connect-docker #13: docker pull --> no space left on device

amamus, 2022-03-08:
Hi Tony,
thank you for your awesome work!

I would like to share and issue with you, I had to deal with.
I ran into "no space left on device" while pulling docker since default docker path is /var/lib/docker/ and default partion size is too small.

Solution is actually pretty easy:
`mkdir /data/docker_storage
chmod -R 777 /data/docker_storage` 

Change /usr/lib/systemd/system/docker.service
from: 
`ExecStart=/usr/bin/dockerd -H fd://`
to:
`ExecStart=/usr/bin/dockerd -g /data/docker_storage -H fd://`

mv /var/lib/docker/* /data/docker_storage/.

reboot hifiberry

check your result:
# ps aux | grep -i docker | grep -v grep

and start tidal docker installation again.

I hope this issue is a benefit to some folks.

Take care
David

kmare2, 2022-05-08:
thank you!!

docker/plugin working well with this workaround

pfbreton-perso, 2022-06-08:
Hi!  I am not familiar enough with ssh and linux to understand how to perform theses steps:

------------
Change /usr/lib/systemd/system/docker.service
from:
ExecStart=/usr/bin/dockerd -H fd://
to:
ExecStart=/usr/bin/dockerd -g /data/docker_storage -H fd://

mv /var/lib/docker/* /data/docker_storage/.
------------------------

can you provide a bit more details to newbies like me?

amamus, 2022-06-12:
change file content:
1. connect via ssh to hifiberrry
2. vi /usr/lib/systemd/system/docker.service (please google for usage)
3. look out for row which needs needs to be chaged "ExecStart ..."
4. change it
5. save the file.

move files:
1. connect via ssh to hifiberrry
2. mv /var/lib/docker/* /data/docker_storage/.

That's it actually.
Note: 
mv == shell move command
vi is a shell editor

pfbreton-perso, 2022-06-17:
Thanks for the pointers!  I was able to figure it out with those extra hints!   I also found that managing most of this with WinSCP is a lot easier than using a terminal :)

RalpinMcDjango, 2022-10-22:
> Hi Tony, thank you for your awesome work!
>
> I would like to share and issue with you, I had to deal with. I ran into "no space left on device" while pulling docker since default docker path is [...]

Thanks for sharing this fix David