TonyTromp/tidal-connect-docker #14: Music not playing - Content-Range is missing from response / Timeout was reached

mkScript, 2022-03-13:
Hi Tony,

Thanks for your efforts for that great project. I use it regularly with hifiberryOS in combination with the DAC 2 HD since a year now. Until last week I didn't face any issues. But now, without any updates of hifiberryOS / tidal-connect-docker I recently face the issue that music is not going to play. Here's the log output, which is pretty similar to the one of Dave in the HifiBerry forum (https://support.hifiberry.com/hc/en-us/community/posts/360013667717/comments/4472867272337):

```
Starting Speaker Application in Background (TMUX)
Starting TIDAL Connect..
STARTING TidalConnect services, version: 1.1.3
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2022-03-13 19:41:04.274] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2022-03-13 19:41:23.320] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5516 milliseconds
[2022-03-13 19:41:23.320] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:41:30.257] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
[2022-03-13 19:41:30.257] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:41:35.792] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
[2022-03-13 19:41:35.792] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:41:41.326] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5516 milliseconds
[2022-03-13 19:41:41.326] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:41:41.329] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
[2022-03-13 19:41:47.477] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:0
[2022-03-13 19:41:47] [info] asio async_shutdown error: asio.misc:2 (End of file)
[2022-03-13 19:42:11.808] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2022-03-13 19:42:29.770] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
[2022-03-13 19:42:29.770] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:42:35.298] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5516 milliseconds
[2022-03-13 19:42:35.298] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:42:40.823] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5513 milliseconds
[2022-03-13 19:42:40.824] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:42:40.825] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
[2022-03-13 19:42:44] [info] asio async_shutdown error: asio.misc:2 (End of file)
[2022-03-13 19:43:41.229] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2022-03-13 19:43:59.166] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
[2022-03-13 19:43:59.166] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:44:04.696] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
[2022-03-13 19:44:04.696] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:44:10.223] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
[2022-03-13 19:44:10.223] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-13 19:44:10.224] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
```

I also updated hifiberryOS and your project to the newest version which did not help.

When I run the script to get the DAC's name it returns the following:

`snd_rpi_hifiberry_dacplushd: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 (hw:0,0)`

As I have the DAC 2 HD (!=  DAC+ HD) installed I was wondering whether this might also the reason for the issue. 

Best
Michael

Python21212, 2022-03-27:
I have a similar issue (DAC2 Pro displaying as DAC+ Pro on HifiBerry OS).

chsymann, 2022-03-28:
Same here using DIGI+ PRO. 
```
[2022-03-28 10:52:58.203] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
[2022-03-28 10:53:04.019] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
[2022-03-28 10:53:04.020] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-28 10:53:09.548] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
[2022-03-28 10:53:09.549] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-28 10:53:15.079] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
[2022-03-28 10:53:15.079] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2022-03-28 10:53:15.083] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
[2022-03-28 10:53:40.210] [tisoc] [error] [SpkconServer.cpp:368] recv error. socket disconnected
```

exislow, 2023-02-16:
I have a super strange error running TIDAL Connect Docker on my HiFiBerry. After the installation everything worked as expected for month. Then I switched the HiFiBerry off for two weeks and afterwards the Docker container isn't really working anymore. I even tried to re-install it. The docker logs showing this error:

```
# docker-compose up
tidal_connect is up-to-date
Attaching to tidal_connect
tidal_connect    | Starting Speaker Application in Background (TMUX)
tidal_connect    | Starting TIDAL Connect..
tidal_connect    | STARTING TidalConnect services, version: 1.1.3
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | Cannot connect to server socket err = No such file or directory
tidal_connect    | Cannot connect to server request channel
tidal_connect    | jack server is not running or cannot be started
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | tidal_connect_application: simple-watch.c:99: avahi_simple_poll_wakeup: Assertion `s' failed.
tidal_connect    | /entrypoint.sh: line 19:    12 Aborted                 (core dumped) /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path "/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" --playback-device "${PLAYBACK_DEVICE}" -f "${FRIENDLY_NAME}" --codec-mpegh true --codec-mqa ${MQA_CODEC} --model-name "${MODEL_NAME}" --disable-app-security false --disable-web-security false --enable-mqa-passthrough ${MQA_PASSTHROUGH} --log-level 3 --enable-websocket-log "0"
tidal_connect    | TIDAL Connect Container Stopped..
tidal_connect    | Starting Speaker Application in Background (TMUX)
tidal_connect    | Starting TIDAL Connect..
tidal_connect    | STARTING TidalConnect services, version: 1.1.3
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | Cannot connect to server socket err = No such file or directory
tidal_connect    | Cannot connect to server request channel
tidal_connect    | jack server is not running or cannot be started
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[... cut ...]
tidal_connect    | [2023-02-16 09:15:53.740] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal_connect    | [2023-02-16 09:15:53.743] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
tidal_connect    | [2023-02-16 09:16:27] [info] asio async_shutdown error: asio.misc:2 (End of file)
```

Music tries to be streamed but I cannot hear something. And as soon `tidal_connect    | [2023-02-16 09:16:27] [info] asio async_shutdown error: asio.misc:2 (End of file)` appears my TIDAL app disconnect from the TIDAL Connect server.

HiFiBerry OS: 20221128