TonyTromp/tidal-connect-docker #17: Raspbian OS guide

Python21212, 2022-03-27:
Hi there, how do I install this software on Raspberry Pi OS? It does not seem particularly obvious - all the info is for HiFiBerry OS (which doesn't seem to work well with my Pi for reasons which elude me). I can't find the Docker compose scripts in the Docker folder even after downloading (although I wouldn't know what to do with them anyway :) ).

Thanks
Monty

charliesjc, 2022-03-27:
I'm running DietPi on a RPi4B and I couldn't be bothered to try and "exclude" the HifiBerry OS bits. In the end I just ran ./install_hifiberry.sh after installing docker and docker-compose. Everything works perfectly. There are some errors but they all seem to be with the beocreate server stuff, because obviously there is no such thing on DietPi.

TonyTromp, 2023-02-14:
This is great to know. I have never tried DietPi, Thanks!

closed 2023-02-14 by TonyTromp