TonyTromp/tidal-connect-docker #19: Tidal Connect | MQA not working after rebooting the hifiberry OS.

Pikkuv, 2022-03-27:
Dear Tony,

I have done changes in the CONFIG file but still the MQA not working. It didnt work after reboot the hifiberry os.

Here are my CONFIG file and docker-compose.yml file 

CONFIG file 
FRIENDLY_NAME=hifiberry
MODEL_NAME=hifiberry
MQA_PASSTHROUGH=True
MQA_CODEC=True
PLAYBACK_DEVICE=EarMen TR-AMP: USB Audio (hw:1,0)

docker-compose.yml 
version: '2.2'
services:
  tidal-connect:
    container_name: tidal_connect
    env_file:
    - .env
    image: edgecrush3r/tidal-connect:latest
    tty: true
    restart: always
    network_mode: host
    devices:
     - /dev/snd
    volumes:
      - ./entrypoint.sh:/entrypoint.sh
      - /var/run/dbus:/var/run/dbus
    restart: always
    dns:
      - 8.8.8.8

Thank you

regards

charliesjc, 2022-03-30:
I don't think you can use passthrough MQA on Hifiberry OS. (You can't send unfolded MQA down an SPDIF/Toslink pipe, it has to go via USB, which Hifiberry doesn't do.)The best you'll get is the 1st unfold using the MQA codec. I know the option is there to turn passthrough on but this whole thing wasn't designed specifically for Hifiberry in the first place.

TonyTromp, 2022-03-31:
MQA passthrough is working for some. xan you try
MQA_PASSTHROUGH=True
MQA_CODEC=False

charliesjc, 2022-04-01:
I don't think you can have USB out on anything running HifiBerry OS. Unless there's a way to "hack" it a bit. And as far as I know MQA passthrough has to be via USB.

charliesjc, 2022-04-01:
Nevermind. I've just done it on HifiBerry OS. Running MQA passthrough on an Ifi Zen DAC V2. I don't own a DAC HAT from hifiberry.

charliesjc, 2022-05-04:
Found my way back to this thread while looking for updates: Your EarMen TR-AMP only supports MQA Rendering so you need specify `MQA_PASSTHROUGH=false` as the first unfold unfold needs to be performed by the tidal service. Your device can perform the "final unfold", not the whole thing. IF you have a light indicating something it should probably be magenta.

TonyTromp, 2022-05-04:
That is great news. Unfortunately i dont have a eternal DAC to test this, nor contribute.

charliesjc, 2022-05-04:
Let me know if there's anything you want me to test. I use an Ifi Zen DAC V2 and I've also purchased a DAC2 Pro (but I'll only see that in June when my in-laws arrive from the US). My Zen DAC V2 produces a Magenta light if I set the passthrough to false: it then runs in "render mode" instead of "decode and render". It produces green or blue as is the case if passthrough is set to true.

After running the --help function on the tidal_connect_application I think the --codec-mqa doesn't actually modify anything going to the DAC per se. It seems to tell the connect application whether to make MQA content available or not (but it doesn't seem to make any difference to what I can and can't play on my side). The --codec-mpegh seems to be related to the 360 headphone content. And there is a Dolby Atmos codec mention in the --help as well. I actually found the details first by opening the application in a hex editor and simply searching for "--" to see the plaintext and later just ran the app using the --help flag.