TonyTromp/tidal-connect-docker #21: HDMI audio not working

imp1sh, 2022-06-02:
When I choose HDMI audio output during installation wizard it won't work.
The Tidal App seems to be connected but the track just isn't playing, it's stuck at the given time.

This is an excerpt from the logs:
```
{"log":"ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.436486321Z"}
{"log":"ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.436595581Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958\r\n","stream":"stdout","time":"2022-06-02T23:02:20.436701299Z"}
{"log":"ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'\r\n","stream":"stdout","time":"2022-06-02T23:02:20.436802018Z"}
{"log":"ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.436901227Z"}
{"log":"ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.437273533Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif\r\n","stream":"stdout","time":"2022-06-02T23:02:20.437404926Z"}
{"log":"ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'\r\n","stream":"stdout","time":"2022-06-02T23:02:20.438731407Z"}
{"log":"ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.439310724Z"}
{"log":"ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.439691311Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif\r\n","stream":"stdout","time":"2022-06-02T23:02:20.440062315Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi\r\n","stream":"stdout","time":"2022-06-02T23:02:20.440450557Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi\r\n","stream":"stdout","time":"2022-06-02T23:02:20.440802188Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem\r\n","stream":"stdout","time":"2022-06-02T23:02:20.441173609Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem\r\n","stream":"stdout","time":"2022-06-02T23:02:20.441675954Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline\r\n","stream":"stdout","time":"2022-06-02T23:02:20.44219861Z"}
{"log":"ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline\r\n","stream":"stdout","time":"2022-06-02T23:02:20.442669552Z"}
{"log":"Cannot connect to server socket err = No such file or directory\r\n","stream":"stdout","time":"2022-06-02T23:02:20.499089077Z"}
{"log":"Cannot connect to server request channel\r\n","stream":"stdout","time":"2022-06-02T23:02:20.499364518Z"}
{"log":"jack server is not running or cannot be started\r\n","stream":"stdout","time":"2022-06-02T23:02:20.50182927Z"}
{"log":"JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock\r\n","stream":"stdout","time":"2022-06-02T23:02:20.502592734Z"}
{"log":"JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock\r\n","stream":"stdout","time":"2022-06-02T23:02:20.503264125Z"}
{"log":"STARTED TidalConnect services.\r\n","stream":"stdout","time":"2022-06-02T23:02:20.671129402Z"}
{"log":"[2022-06-02 23:02:40.509] [tisoc] [\u001b[33m\u001b[1mwarning\u001b[m] [PlaybackControllerImpl.cpp:472] Illegal state:1\r\n","stream":"stdout","time":"2022-06-02T23:02:40.510702454Z"}
{"log":"[2022-06-02 23:02:41.386] [tisoc] [\u001b[33m\u001b[1mwarning\u001b[m] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!\r\n","stream":"stdout","time":"2022-06-02T23:02:41.387738439Z"}
{"log":"Expression 'hostSampleFormat = PaUtil_SelectClosestAvailableFormat( GetAvailableFormats( self-\u003epcm ), userSampleFormat )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1910\r\n","stream":"stdout","time":"2022-06-02T23:02:41.589192112Z"}
{"log":"Expression 'PaAlsaStreamComponent_Initialize( \u0026self-\u003eplayback, alsaApi, outParams, StreamDirection_Out, NULL != callback )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2175\r\n","stream":"stdout","time":"2022-06-02T23:02:41.589588092Z"}
{"log":"Expression 'PaAlsaStream_Initialize( stream, alsaHostApi, inputParameters, outputParameters, sampleRate, framesPerBuffer, callback, streamFlags, userData )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2840\r\n","stream":"stdout","time":"2022-06-02T23:02:41.589707985Z"}
{"log":"Could not start stream-9988: Invalid stream pointer\r\n","stream":"stdout","time":"2022-06-02T23:02:41.590437445Z"}
{"log":"[2022-06-02 23:04:10] [info] asio async_shutdown error: asio.misc:2 (End of file)\r\n","stream":"stdout","time":"2022-06-02T23:04:10.153853526Z"}
{"log":"[2022-06-02 23:04:17.365] [tisoc] [\u001b[31m\u001b[1merror\u001b[m] [SpkconServer.cpp:368] recv error. socket disconnected\r\n","stream":"stdout","time":"2022-06-02T23:04:17.366888788Z"}
```

This is the config file:
```
FRIENDLY_NAME=raspberrypi
MODEL_NAME=raspberrypi
MQA_PASSTHROUGH=false
MQA_CODEC=false
PLAYBACK_DEVICE=vc4-hdmi: MAI PCM i2s-hifi-0 (hw:1,0)
```

Is it possible that HDMI audio output just isn't possible?
This is not hifiberry OS but Raspbian 11.3.

TonyTromp, 2023-02-14:
Do you get any other Audio via HDMI working? On Hifiberry this is not possible (not supported by OS).

npmstart-pray, 2023-02-14:
> Do you get any other Audio via HDMI working? On Hifiberry this is not possible (not supported by OS).

This is tragic. Doesn't HDMI mean High Definition Media Interface?

TonyTromp, 2023-02-15:
I think there is a way via the ALSA mixer to sort this out. You could try "default" or "sysdedefault" from installation menu. Alternatively Instead of running the process througb docker (which would require you to passthrough your HDMI device, you can also try and run the binary directly on RaspbianOS. Have a look at the entrypount.sh script for the command line options.