TonyTromp/tidal-connect-docker #23: Can't select soundcard

off112a, 2022-06-08:
Hi, I have a Pi 0W with a HiFiBerry DAC+ Zero. I can't select a soundcard when I run the ./install_hifiberry.sh.

I have also tried 
`docker run --device /dev/snd \
  -v /var/run/dbus:/var/run/dbus \
  -v /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket \
  --entrypoint "" \
  edgecrush3r/tidal-connect /app/ifi-tidal-release/bin/ifi-pa-devs-get 2>/dev/null | grep device#`

But it shows nothing.

I have also tried to manually adding info to the entrypoint.sh file. But it only gives error when i try to start tidal service.

Log:

tidal_connect    | Starting Speaker Application in Background (TMUX)
tidal_connect    | Starting TIDAL Connect..
tidal_connect    | /entrypoint.sh: line 19:    12 Illegal instruction     (core dumped) /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path "/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" --playback-device "snd_rpi_hifiberry_dac: HifiBerry DAC HiFi pcm5102a-hifi-0 (hw:0,0)" -f "HiFiBerry" --codec-mpegh true --codec-mqa false --model-name "HiFiBerry" --disable-app-security false --disable-web-security false --enable-mqa-passthrough false --log-level 3 --enable-websocket-log "0"
tidal_connect    | TIDAL Connect Container Stopped..

`

jmayday, 2022-12-08:
Same issue here. My ADI DAC-2 FS is not listed

```
kuba@raspberrypi:~/tidal-connect-docker $ sudo ./install_hifiberry.sh
Running environment:
  FRIENDLY_NAME:            raspberrypi
  MODEL_NAME:               raspberrypi
  BEOCREATE_SYMLINK_FOLDER: /opt/beocreate/beo-extensions/tidal
  DOCKER_DNS:               8.8.8.8
  DOCKER_IMAGE:             edgecrush3r/tidal-connect:latest
  BUILD_OR_PULL:            pull
  MQA_PASSTHROUGH:          false
  MQA_CODEC:                false
  PWD:                      /home/kuba/tidal-connect-docker

Thu 08 Dec 2022 10:23:32 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Pre-flight checks.
Thu 08 Dec 2022 10:23:32 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Checking to see if Docker is running.
Thu 08 Dec 2022 10:23:32 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Confirmed that Docker daemon is running.
Thu 08 Dec 2022 10:23:32 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Checking to see if Docker image edgecrush3r/tidal-connect:latest exists.
Thu 08 Dec 2022 10:23:32 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Docker image edgecrush3r/tidal-connect:latest exist on the local machine.
Thu 08 Dec 2022 10:23:32 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Stopping Tidal Container..

Stopping TIDAL Connect Service.
Finished stopping TIDAL Connect Service.
Thu 08 Dec 2022 10:23:44 PM CET raspberrypi install_hifiberry.sh[1894]: [INFO]: Select audio output device

Found output devices...

0=sysdefault
1=default
2=dmix
Choose your output Device (0-9): 1
```

```
kuba@raspberrypi:~/tidal-connect-docker $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
kuba@raspberrypi:~/tidal-connect-docker $ docker run --device /dev/snd   -v /var/run/dbus:/var/run/dbus   -v /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket   --entrypoint ""   edgecrush3r/tidal-connect /app/ifi-tidal-release/bin/ifi-pa-devs-get 2>/dev/null | grep device#
kuba@raspberrypi:~/tidal-connect-docker $
```

bodnaristvan, 2023-01-22:
Hey @jmayday, i was having a similar issue, and realized it's because i had a 64bit OS installed, while docker trying to run something in 32bit mode. Or at least that's my guess..

This was the error message that helped me:
> WARNING: The requested image's platform (linux/arm) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested

I'm sure there's a simple solution for this just using Docker, rebuilding an image, choosing a different base image, IDK. For me, it was easier to just start fresh, with a 32bit base OS instead.