TonyTromp/tidal-connect-docker #25: 'Now Playing' does not display song information

serveturan, 2022-08-10:
Hi Tony,

After successful installation I can now play via Tidal Connect but I can't see the song information at 'Now Playing' display in Hifiberry Os page. I only see  the song information from Spotify Connect which is played before Tidal Connect. It stuck there or sometimes I see nothing. Is it known issue? I'm using Pi 4 +Hifiberry DAC2pro + Hifiberry DSP board

Thanks in advance
Servet

0cta, 2022-11-15:
Would also embrace this feature but I'm afraid this would require quite some extra work. I'm not even sure how HifiBerry OS is polling that information from Spotify Connect or other sources, might be MPRIS?

TonyTromp, 2023-02-11:
This feature has not been officially released yet, but i have put my efforts in the work-in-progress folder.

closed 2023-02-14 by TonyTromp