TonyTromp/tidal-connect-docker #28: Tidal APK 2.65.1 - [error] handle_transport_init received error: TLS handshake failed

paradix, 2022-08-17:
After updating to APK 2.65.1 tidal connect stopped working.

Docker starts but the you can't connect to the player from the Android phone.
The following error occurs:
[error] handle_transport_init received error: TLS handshake failed

Can anyone confirm?

MoM-Raider, 2022-08-18:
I can confirm this. It was also reported in another Thread. I fear they revoked the certificate that is used by the ifi leak we are using but understand to little about the handshake to be certain

SebastianoCurreli, 2022-08-19:
I can confirm this too. 
For the sake of leaving an helpful note to the others encountering this issue: 
The last working APK is version 2.64.2, thus reverting is a temporary workaround.
@MoM-Raider can you link the thread where they report this issue?

SebastianoCurreli, 2022-08-19:
@TonyTromp in case @MoM-Raider 's guess would be correct (currently used ifi certificate being revoked) can you please tell us  which one of the certs available in the [seniorgod repo](https://github.com/seniorgod/ifi-tidal-release/tree/master/id_certificate) is currently the default one? We could run tests using the other one.

paradix, 2022-08-20:
@SebastianoCurreli based on the entrypoint.sh script the certificate is used
`app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat`

SebastianoCurreli, 2022-08-20:
@paradix I've tested both the certificates available at [seiorgod repo](https://github.com/seniorgod/ifi-tidal-release/tree/master/id_certificate) on Tidal APKs version  2.64.2 and 2.65.1 and Windows client version 2.32.0.12.
On Android devices both certificate work up to version 2.64.2,  from version 2.65.1 both certificates failed the handshake.
It seems that  @MoM-Raider is right, either certificates got revocated or linked device IDs got blacklisted, yet they fail to handshake.
On Windows, both certificate work fine up to 2.32.0.12 (latest for Windows 10), yet I'd not be surprised to see the block implemented for the windows client too in few days.
This sucks. 
As of today blocking updates seems the only viable workaround.

charliesjc, 2022-08-23:
Tidal seems to have made a mess of this anyway. I'm also using Volumio which is a Tidal sanctioned implementation of Tidal Connect and since a couple of days ago it has also broken with v2.65.1. On v2.65.0 and below it was still okay.

SebastianoCurreli, 2022-08-23:
Maybe we could advise users to make a mass signalling to Tidal (i.e. on Reddit or to Tidal itself) to prompt Tidal devs to fix the mess.

TonyTromp, 2022-08-26:
I truly hope Tidal opens up to the opensource community, and to be honest i have already moved on (away from using Tidal). Its a sad day.

SebastianoCurreli, 2022-08-26:
@TonyTromp what service are you using if I can ask? Do you have another open source project related to hi-res music streaming?

mtimms1980, 2022-08-26:
It is frustrating, I only started using Tidal a few days before this happened and was stoked that this existed! 

Oddly old versions of the app don't show the tidal connect service, and the old version of the app is laggy on my Pixel 6. 

I was deciding between Qubuz and Tidal, and went for Tidal as it does offer some level of curated recommendations which I quite enjoyed on Spotify. As far as I am aware Qobuz does not do this. 

Do you think this is an active move by Tidal to stop projects like this?

npmstart-pray, 2022-08-28:
Possibly, but I'd lean more towards saying it's for their own internal purposes.

TonyTromp, 2022-09-08:
> @TonyTromp what service are you using if I can ask? Do you have another open source project related to hi-res music streaming?

I just started to use BluOS in my every day life as this just works and have most features i like in a digital setup. However I do miss the DSP settings from my Pi as this greatly improved overall sound reproduction in my room. That beeing said, i dont think that Tidal will ever provide an opensource API or free license for software like HifiBerry. Thus best is to move away unfortunately.

reidyrobot, 2022-09-21:
I can't see it as a separate logged issue so just to confirm the same problem using the iOS app from my ipad. It's a bit trickier to revert in iOS and I haven't tried yet.

Gee, it'd be so great to have this working again.

paradix, 2022-09-22:
@reidyrobot Can you please post a link to that volumio thread?

charliesjc, 2022-09-22:
This one?

https://community.volumio.org/t/lost-tidal-connect-after-update/57503/28

reidyrobot, 2022-09-22:
> @reidyrobot Can you please post a link to that volumio thread?

https://community.volumio.org/t/tidal-connect-not-working-from-android-phone/57780/2

paradix, 2022-09-22:
Tidal 2.68 still does not work.

reidyrobot, 2022-09-22:
Weirdly enough when I tried to update my Android phone to test, it doesn't install 2.68 but rather an earlier 2.67x version, which as noted doesn't work.

the4anoni, 2022-09-25:
Anyone knows how to block auto updates on desktop Tidal? 2.33.2.611-release works for now

Kybeer, 2022-09-29:
> Tidal 2.68 still does not work.

Same for me

npmstart-pray, 2022-09-30:
Could it be that this is the way:

https://flathub.org/apps/details/com.mastermindzh.tidal-hifi

?

emerout, 2022-10-20:
I use iOs since android app does not work anymore.

On iOs,  I haven't updated Tidal app for several month, to avoid the same issue than with android.

Since a couple of day, iOs Tidal app doesn't work anymore, but without any TLS handshake failure in the tidal_connect log.

So I decided to update the iOs Tidal app, and surprisingly it works again ! 
It seems that they keep the same certificate than before.

paradix, 2023-02-06:
@TonyTromp there is a solution for the handshake issue. Reach out please so that your docker image can be fixed

charliesjc, 2023-02-06:
Ooh. I'm very keen to see what the fix is.

tiscali19, 2023-02-06:
Hello, what is the solution to the problem? I can connect only with Win10 and iOs...

TonyTromp, 2023-02-06:
> @TonyTromp there is a solution for the handshake issue. Reach out please so that your docker image can be fixed

Many many Thanks @paradix. I will test today and hope to have a new build ready by tomorrow! Looking very promising!

mtimms1980, 2023-02-08:
This is great news! How is the testing looking?

mtimms1980, 2023-02-08:
Excellent!

SebastianoCurreli, 2023-02-08:
If you guys need another tester feel free to reach out. I'm on HiFiBerryOS 221128, RPi4B Rev 1.2

charliesjc, 2023-02-08:
Great, can’t wait to see how it was done.

Kybeer, 2023-02-10:
> > @TonyTromp there is a solution for the handshake issue. Reach out please so that your docker image can be fixed
> 
> Many many Thanks @paradix. I will test today and hope to have a new build ready by tomorrow! Looking very promising!

Hi, any news to share? 😁😁😄

TonyTromp, 2023-02-10:
Hi everyone,

Great news! I have patched the files with the certificate provided by @paradix. If people are willing to test together with me, i have checked in the latest version in this branch: https://github.com/TonyTromp/tidal-connect-docker/tree/bug/issue-28_tidal-apk-TLS-handshake. I will need to re-install my Hifiberry (as it was gathering dust for last month), so sorry if testing has been slow.

1. Switch the 'Master' branch into '[bug/issue-28_tidal-apk-TLS-handshake]'
2. Copy these files to your Raspberry Pi
3. From terminal 'cd src/Docker' and build the docker image using './build_docker.sh' script
4. run the 'install_hifiberry.sh' script

Happy testing!!! + Special thanks to @paradix for making us very Happy!!

Kybeer, 2023-02-10:
> Hi everyone,
>
> Great news! I have patched the files with the certificate provided by @paradix. If people are willing to test together with me, i have checked in the latest version in this branch: [...]

It's working. Thanks a lot for the effort!

SebastianoCurreli, 2023-02-10:
Great news, I'll test it tomorrow!
Thanks @TonyTromp @paradix

tiscali19, 2023-02-10:
Thanks @TonyTromp and @paradix, It's working on my Android device!

P.S. @SebastianoCurreli , vivi in Sardegna?

TonyTromp, 2023-02-11:
Great! With this latest effort, development on this project for HifiBerry can resume again. I will merge changes to master later today and checkin latest images on docker.io

SebastianoCurreli, 2023-02-11:
tldr: @TonyTromp  @paradix  I've just tried the new certificate and it doesn't work on my rig with the latest Tidal APK.

Anybody testing had a similar outcome?
@paradix @Kybeer @tiscali19  what's the Tidal APK version you guys are using?

Details:
For the sake of details I'm posting all the steps to check whether I've made some trivial mistake.

1. Rename the "old" certificate  "IfiAudio_ZenStream.dat" to "IfiAudio_ZenStream.expired"
2. Dowload the new certificate "IfiAudio_ZenStream.dat" from [https://github.com/TonyTromp/tidal-connect-docker/tree/bug/issue-28_tidal-apk-TLS-handshake/Docker/src/id_certificate/](url)
3. rebuild the docker image using the "build_docker.sh" script
4. installed the plugin using "install_hifiberry.sh' "script

I am on HiFiBerryOS 221128, RPi4B Rev 1.2
Tidal APK is the latest on Android 2.75.0.1096.9546

The new certificate works if I revert to Tidal APK 2.64.2

paradix, 2023-02-11:
I'm using Tidal APK 2.75.0 without issues on two different setups. Works without issues.

You must have messed up something with the image build process.

Kybeer, 2023-02-11:
> Anybody testing had a similar outcome?
> @paradix @Kybeer @tiscali19 what's the Tidal APK version you guys are using?

@SebastianoCurreli I have no issues with this setup;

Raspberrypi 3 + RapsberryOS 11 + Docker
Android phone + Tidal app v. 2.76.0
Installed Tidal Connect using instructions above from @TonyTromp

SebastianoCurreli, 2023-02-11:
> tldr: @TonyTromp @paradix I've just tried the new certificate and it doesn't work on my rig with the latest Tidal APK.
>
> Anybody testing had a similar outcome? @paradix @Kybeer @tiscali19 what's the [...]

Issue fixed. 
For some reason my "./Docker/entrypoint.sh" file was pointing to "IfiAudio_NeoStream.dat" certificate. 
Updating the reference to the new "IfiAudio_ZenStream.dat" certificate solved the issue. 

Now I can confirm that HiFiBerryOS 221128 on RPi4B Rev 1.2 and Tidal APK 2.75.0.1096.9546 (latest on Android) have a working TidalConnect service.

Many thanks to @TonyTromp @paradix

Kybeer, 2023-02-11:
> > @TonyTromp there is a solution for the handshake issue. Reach out please so that your docker image can be fixed
> 
> Many many Thanks @paradix. I will test today and hope to have a new build ready by tomorrow! Looking very promising!

Will this new certificate eventually expire (I assume yes?) and create the same problem again?

npmstart-pray, 2023-02-11:
Expire or be revoked, yes, of course. That’s how it works by design. Sent from nowhere near my ready room with my new tricorder

TonyTromp, 2023-02-11:
Is the new build with certificate and instructions not working for you?

TonyTromp, 2023-02-11:
If its similar to x509, We should be able to check the validity of the certificate i hope

mtimms1980, 2023-02-12:
> Hi everyone,
>
> Great news! I have patched the files with the certificate provided by @paradix. If people are willing to test together with me, i have checked in the latest version in this branch: [...]

Sorry if I am being an idiot, I have en existing pi with hifiberry setup from when this worked previously. What would the steps be to apply this fix? Or even from a newly setup pi using hifiberry. Thanks in advance.

tiscali19, 2023-02-12:
> 
@mtimms1980 : you are using Docker? If not using, download new IfiAudio_ZenStream.dat file and overwite them

mtimms1980, 2023-02-12:
@tiscali19 I am using Docker, but only because I started from TonyTromp's method. Happy to start from scratch to get this up and running again.

mtimms1980, 2023-02-12:
Just started from scratch, and it is working great! (some bass distortion, but I think that is probably my receiver setup!) Thanks for the hard work on this.

TonyTromp, 2023-02-13:
I have merged the Pull Request with the main branch.

If there are any issues, please create a new 'issue' and we can follow-up (as this issue is auto-closed by the PR merge).

closed 2023-02-13 by TonyTromp

reidyrobot, 2023-03-18:
Just dropped by to say thanks for sorting this issue! much appreciated.