TonyTromp/tidal-connect-docker #29: Docker not working with android devices

TomMar71, 2022-08-22:
The installation seems to have worked fine as Tidal connect works as intended from my Windows PC, however when I try and select the Device (which does show up) on my Android 10 Tablet it just stops playback. When I press resume it automatically jumps back to the Tablet speakers.

TomMar71, 2022-08-22:
just read the previous issue thread, probably the same issue

Kybeer, 2022-09-28:
Same issues here

dzavux, 2022-10-18:
It did not lasted very long. This mourning I was able to use Tidal connect, Later it stopped. I can still find Tidal connect end point, but no sound.

RalpinMcDjango, 2022-10-22:
I also have this issue running Android 12 and the Official Tidal app.

The device will appear in the list of available devices then once you select it fails to connect instantly and "disconnects" the stream.

TonyTromp, 2023-02-10:
This is related to this error: https://github.com/TonyTromp/tidal-connect-docker/issues/28

closed 2023-02-14 by TonyTromp