TonyTromp/tidal-connect-docker #32: Music not playing - Content-Range is missing from response / Timeout was reached

exislow, 2023-04-09:
I have a super strange error running TIDAL Connect Docker on my HiFiBerry. After the installation everything worked as expected for month. Then I switched the HiFiBerry off for two weeks and afterwards the Docker container isn't really working anymore. I even tried to re-install it. The docker logs showing this error:

```
# docker-compose up
tidal_connect is up-to-date
Attaching to tidal_connect
tidal_connect    | Starting Speaker Application in Background (TMUX)
tidal_connect    | Starting TIDAL Connect..
tidal_connect    | STARTING TidalConnect services, version: 1.1.3
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | Cannot connect to server socket err = No such file or directory
tidal_connect    | Cannot connect to server request channel
tidal_connect    | jack server is not running or cannot be started
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | tidal_connect_application: simple-watch.c:99: avahi_simple_poll_wakeup: Assertion `s' failed.
tidal_connect    | /entrypoint.sh: line 19:    12 Aborted                 (core dumped) /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path "/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" --playback-device "${PLAYBACK_DEVICE}" -f "${FRIENDLY_NAME}" --codec-mpegh true --codec-mqa ${MQA_CODEC} --model-name "${MODEL_NAME}" --disable-app-security false --disable-web-security false --enable-mqa-passthrough ${MQA_PASSTHROUGH} --log-level 3 --enable-websocket-log "0"
tidal_connect    | TIDAL Connect Container Stopped..
tidal_connect    | Starting Speaker Application in Background (TMUX)
tidal_connect    | Starting TIDAL Connect..
tidal_connect    | STARTING TidalConnect services, version: 1.1.3
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal_connect    | Cannot connect to server socket err = No such file or directory
tidal_connect    | Cannot connect to server request channel
tidal_connect    | jack server is not running or cannot be started
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal_connect    | /entrypoint.sh: line 19:    12 Segmentation fault      (core dumped) /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path "/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" --playback-device "${PLAYBACK_DEVICE}" -f "${FRIENDLY_NAME}" --codec-mpegh true --codec-mqa ${MQA_CODEC} --model-name "${MODEL_NAME}" --disable-app-security false --disable-web-security false --enable-mqa-passthrough ${MQA_PASSTHROUGH} --log-level 3 --enable-websocket-log "0"
tidal_connect    | TIDAL Connect Container Stopped..
tidal_connect    | Starting Speaker Application in Background (TMUX)
tidal_connect    | Starting TIDAL Connect..
tidal_connect    | STARTING TidalConnect services, version: 1.1.3
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal_connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
[... 50 lines cut ...]
# cat .env
FRIENDLY_NAME=hifiberry
MODEL_NAME=hifiberry
MQA_PASSTHROUGH=true
MQA_CODEC=false
PLAYBACK_DEVICE=snd_rpi_hifiberry_digi: HifiBerry Digi HiFi wm8804-spdif-0 (hw:0,0)
```

What is going on here? As soon as I connect via TIDAL to my TIDAL Connect server all the errors appear:

```
tidal_connect    | [2023-02-16 09:15:24.919] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal_connect    | [2023-02-16 09:15:42.681] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal_connect    | [2023-02-16 09:15:42.681] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal_connect    | [2023-02-16 09:15:48.211] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal_connect    | [2023-02-16 09:15:48.212] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal_connect    | [2023-02-16 09:15:53.740] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5513 milliseconds
tidal_connect    | [2023-02-16 09:15:53.740] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal_connect    | [2023-02-16 09:15:53.743] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
tidal_connect    | [2023-02-16 09:16:27] [info] asio async_shutdown error: asio.misc:2 (End of file)
```

Music tries to be streamed but I cannot hear something. And as soon `tidal_connect    | [2023-02-16 09:16:27] [info] asio async_shutdown error: asio.misc:2 (End of file)` appears my TIDAL app disconnect from the TIDAL Connect server.

HiFiBerry OS: 20221128

FelizCoder, 2023-04-13:
I have the same issue on Raspibian OS versions full & lite.

JSH32, 2024-08-18:
@TonyTromp