TonyTromp/tidal-connect-docker #36: APK 2.87.1 breaks Tidal-Connect

SebastianoCurreli, 2023-08-17:
Hi everyone, 
It seems we might be facing the usual certificate issue (similar to the previous issue #28 closed) as updating to APK 2.87.1 broke the tidal connect functionality on my Android Phone.
Can anyone confirm this?

@TonyTromp @paradix I'm available for testing.

Useful note APK 2.86.1 still works for me.

paradix, 2023-08-18:
@SebastianoCurreli I'll have to check that. Didn't notice any change, but also haven't used connect for a while

paradix, 2023-08-18:
@SebastianoCurreli for me it works well at the moment. No issues. Maybe anyone else can check also

SebastianoCurreli, 2023-08-18:
@paradix thanks for testing, which Tidal-APK version are you using?

paradix, 2023-08-18:
I use 2.87.1

TonyTromp, 2023-08-18:
Noted. Hope its not a sign of uet another cert change.

SebastianoCurreli, 2023-08-18:
> I use 2.87.1

Ok, I'll test on multiple devices and post the results here.

SebastianoCurreli, 2023-08-19:
I've tested the latest three APK versions available thay are comaptible with my mobile devices:
- 2.86.1 works
- 2.87.1 doesn't work
- 2.88.1 works
It looks like it might be some issue on the Tidal devs.
@TonyTromp with these circumstances I'd close the issue.
Thanks everyone for testing :)

TonyTromp, 2023-08-20:
Very cool!! Thanks for testing all of these version. Seems that it was a flaky release. My main worry is that at some time they will expire the cert and there is no official integration with Tidal yet.

javixuwi, 2023-11-16:
It seems not working with latest tidal version 2.94.0. it doesn't find tidal connect

paradix, 2023-11-16:
Works just fine over here

SebastianoCurreli, 2023-11-16:
It works even over here, platform Android.
@javixuwi try forcing app stop and clear cache.

javixuwi, 2023-11-17:
> It works even over here, platform Android. @javixuwi try forcing app stop and clear cache.

Tested. It works, thank you for the advise. Great job