TonyTromp/tidal-connect-docker #40: Music is not playing

Vermuden, 2023-09-17:
Hi, I was very happy to find this project.

The installation was successful, and I can connect to the used Raspberry Pi (running the current HifiberryOS) in my tidal app with Tidal Connect.
Unfortunately, when I select a song, the app is not starting it. The played time just remains at 00:00. Occasionally, after waiting for some time, the song does start, but only for around 15 seconds, then the played time returns to 00:00 and nothing happens.

I am using the Tidal app on an iPhone, app version 2.90.1 and on my desktop, app version 2.34.3.26. 
I am using Raspberry Pi 4 b, with 4Gb of RAM with a Hifiberry DAC2 HD. 
The Pi is connected via Ethernet.
Reinstallation of HifiberryOS and the Docker image did not help.

Is this a problem with the Tidal app or, did I mess up the installation?

npmstart-pray, 2023-09-20:
Have you tried AirPlay?

Vermuden, 2023-09-20:
AirPlay works fine, but I will not be using an iPhone for much longer. That is the reason, why I want to use Tidal Connect.

miguelpissarra, 2024-02-20:
+1 with Tidal Connect problems (RPI3b+ with External DAC)
Start happening after the my last update of Tidal App on iPhone and iPad. Tidal App is version 2.101.0 build 5991 on both.
It works fine on Windows 10 with app version 2.36.2.54.

miguelpissarra, 2024-02-20:
OK, uninstall and install Tidal app on both iPhone and iPad and it works again on both. I'm confused...