TonyTromp/tidal-connect-docker #42: Playback does not start: Alsa issues?

exislow, 2023-12-16:
So I have a fresh HifiberryOS installation on an RaspberryPi 3b and after following the tidal connect setup I cannot use Tidal Connect. Playback start is not possible. The logs say:

```
# cat CONFIG
FRIENDLY_NAME=hifiberry
MODEL_NAME=hifiberry
MQA_PASSTHROUGH=false
MQA_CODEC=false
PLAYBACK_DEVICE=snd_rpi_hifiberry_digi: HifiBerry Digi HiFi wm8804-spdif-0 (hw:0,0)
DOCKER_DNS=1.1.1.1

# docker logs tidal_connect
Starting Speaker Application in Background (TMUX)
Starting TIDAL Connect..
STARTING TidalConnect services, version: 1.1.3
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2023-12-16 13:35:13.904] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2023-12-16 13:35:14.610] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1514
Expression 'ValidateParameters( outputParameters, hostApi, StreamDirection_Out )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2825
Could not start stream-9988: Invalid stream pointer
```

JSH32, 2024-08-18:
@TonyTromp

TonyTromp, 2024-08-19:
Can you try another Playback device instead of: snd_rpi_hifiberry_digi: HifiBerry Digi HiFi wm8804-spdif-0 (hw:0,0)

What does: 
> aplay -L

list as devices?

JSH32, 2024-08-19:
Within the container or outside of the container? I've switched to trying to use my Schiit Modi+ USB DAC instead of snd_rpi_hifiberry_digi and I get the same result