TonyTromp/tidal-connect-docker #51: HiFi Berry Install error (space on /usr)

BigAirAdair, 2024-09-29:
Hi,  I get an out of space error /usr/device when installing tidal-connect on my hifi.  As a noob trying to figure our how to resize my partition or understand if its something different.

Thanks! - Bill
[image]

Thom61, 2024-10-25:
I had the same error. I found that the installation downloads the docker packages into folder /usr/lib/docker. My SD-card provided enough space on the partition which is mounted on /data. So I first I moved all contents of /usr/lib/docker to the /data partition and created a symbolic link to /usr/lib/docker:
```
mkdir -p /data/lib
mv /var/lib/docker /data/lib
ln -s /data/lib/docker /var/lib/docker

```
This solved the problem on my system.

Klaasemann, 2024-11-17:
I had the same issue.
You may use HifiberryOS 64Bit.

Regards
Klaas

BigAirAdair, 2024-11-17:
That fixed it, thank you!Sent from my iPhone